rust-osdev/x86_64 #438: [Question] Trait for Entry handler function types?

brandonchinn178, 2023-09-23:
I wanted to write a helper that generalized the function to set for an entry, like:
```rust
fn init_entry<F>(entry: Entry<F>, handle_fn: F) {
    ...
}
```

but `Entry.set_handler_fn` is explicitly implemented for each HandlerFunc type separately, so I don't think I can provide an adequate type bound on `F`.

Instead of implementing `Entry.set_handler_fn` explicitly for each HandlerFunc type, why not provide a trait for all HandlerFunc types? e.g.
```rust
trait HandlerFuncType {
    fn to_handler_fn_addr(&self) -> VirtAddr;
}

macro_rules! impl_handler_func_type {
    ($f:ty) => {
        impl HandlerFuncType for $f {
            #[inline]
            fn to_handler_fn_addr(&self) -> x86_64::VirtAddr {
                VirtAddr::new(*self as u64)
            }
        }
    };
}

impl_handler_func_type!(HandlerFunc);
// all other HandlerFuncs
```
And then do
```rust
impl Entry {
    #[inline]
    pub fn set_handler_fn<F: HandlerFuncType>(&mut self, handler: &F) -> &mut EntryOptions {
        unsafe { self.set_handler_addr(handler.to_handler_fn_addr()) }
    }
}
```

Would this work? I think this is backwards-compatible, so I don't see any downsides to making this change. I'm fairly new to Rust, so let me know if I'm missing anything.

Freax13, 2023-09-23:
> I wanted to write a helper that generalized the function to set for an entry, like:
>
> ```rust
> fn init_entry<F>(entry: Entry<F>, handle_fn: F) {
>     ...
> }
> ```
>
> but `Entry.set_handler_fn` is explicitly [...]

As a workaround you could also implement your own trait responsible for setting the value:
```rust
trait HandlerFuncType: Sized {
    fn set(self, entry: &mut Entry<Self>) -> &mut EntryOptions;
}
```

> Instead of implementing `Entry.set_handler_fn` explicitly for each HandlerFunc type, why not provide a trait for all HandlerFunc types? e.g.
>
> ```rust
> trait HandlerFuncType {
>     fn [...]

The `F` generic would have to be on `Entry` and not just `set_handler_fn`, but other than that I think that *should* work.

> I think this is backwards-compatible, so I don't see any downsides to making this change. 

My gut tells me that this will be backwards-compatible, but we'll have to see what the semver-checks CI job says.

I agree that this would be an improvement. Feel free to open a pr.

closed 2023-09-24 by josephlr

brandonchinn178, 2023-09-24:
Works perfectly, thank you!

https://github.com/brandonchinn178/toy-os/commit/1909071505ac8f415445066f00800e771f8137f1#diff-72920538f8714a99c6475c6c76029b8bbd40681d8c93bc9d58f264dd3418df1dR32-R42

PR #439 (2023-09-24): Add HandlerFuncType trait
Fixes #438