rust-osdev/x86_64 #452: `VirtAddr` arithmetic may panic

vincent-tr, 2024-01-14:
Using this code:
```rust
const ADDR: VirtAddr = VirtAddr::new_truncate(0xFFFF_8000_0000_0000);
let previous_page = ADDR - 4096;
```
produce a panic.

Indeed, applying the standard substraction to `ADDR` produce a new address where bits 48..64 are left untouched, and do not match.

Would it be acceptable to construct the substraction result with `new_truncate()` instead of `new()`?

Freax13, 2024-01-14:
No, as of right now, the panic is intended and I don't think we want to change that. If you're fine wrapping and want to get the value `0x0000_7fff_ffff_0000` as the result, you can use `Step::backward(ADDR, 4096)` (nightly only).

vincent-tr, 2024-01-15:
I get the point.

Thanks for the `Step::backward` solution.

closed 2024-01-15 by vincent-tr