rust-osdev/x86_64 #466: Maybe impl `Default` for `InterruptStackFrameValue`?

GZTimeWalker, 2024-03-15:
I need to store the `InterruptStackFrameValue`, which means I need some way to initialize the `InterruptStackFrameValue`.

As the `_reserved1` and `_reserved2` were introduced in https://github.com/rust-osdev/x86_64/pull/263 for v0.15.0, I found no way to initialize it.

Maybe add a `default` to it?

Freax13, 2024-03-15:
> I need to store the `InterruptStackFrameValue`, which means I need some way to initialize the `InterruptStackFrameValue`.
> 
> As the `_reserved1` and `_reserved2` were introduced in #263 for v0.15.0, I found no way to initialize it.
> 
> Maybe add a `default` to it?

I'm unsure if implementing `Default` is the way to go here. Zero-initializing all the fields may produce a value, but one that's not very useful. I agree that we should have a way to construct an `InterruptStackFrameValue`, so perhaps a normal constructor taking all fields as parameters would be a better fit.

GZTimeWalker, 2024-03-15:
> I'm unsure if implementing `Default` is the way to go here. Zero-initializing all the fields may produce a value, but one that's not very useful. I agree that we should have a way to construct an [...]

I agree. I have changed the issue's title.

Freax13, 2024-03-15:
Feel free to open a PR.

closed 2024-03-15 by Freax13