rust-osdev/x86_64 #498: Unifying physical and virtual address types across the rust OS development ecosystem.

jounathaen, 2024-09-20:
Hi,

I think having the semantic differentiation between a `PhysicalAddress` and a `VirtualAddress` type provides a lot of value for OS development. What I think would be even better, is to have a `[Physical|Virtual]Address` type that is coherent across multiple crates AND across different architectures. This could simplify the reuse of code and the compatibility between projects.

Currently, the situation looks as follows:

- [`x86_64` has type definitions for this](https://github.com/rust-osdev/x86_64/blob/master/src/addr.rs), but this can only be used on `x86_64`. These types are apparently widely used in rust-osdev projects (and I use them in our hypervisor as well).
- I have copycat-ed this [into our `aarch64` crate](https://github.com/hermit-os/rust-aarch64/blob/master/src/paging.rs) to have a compatible type for `aarch64`. However, `riscv64` etc. is still not supported. (And that crate did really not receive the required love to be adopted elsewhere (yet)).
- rust-vmm has a similar approach with their [`Address` Trait](https://github.com/rust-vmm/vm-memory/blob/54c67221bdfbc88a38e7671d4ef790f67bde3dee/src/address.rs#L63). They are using it for [`GuestAddress`'es](https://github.com/rust-vmm/vm-memory/blob/54c67221bdfbc88a38e7671d4ef790f67bde3dee/src/guest_memory.rs#L120), which is a use-case for address types I have as well.
- In the Hermit OS, we are actually [using the `x86` `PAddr` and `VAddr` types ](pub use x86::bits64::paging::{PAddr as PhysAddr, VAddr as VirtAddr};) for `x86`, the types from the aforementioned `aarch64` crate for the ARM architecture and a custom type for RISC-V

So I was thinking of moving the addresses into a separate crate that is not `x86_64` specific anymore. The types from that crate could be used and re-exported in the `x86_64` crate without changing any interfaces. I guess it is possible to make it compatible with the `x86` crate's types as well. And as I'm on it, I could also implement rust-vmm's traits and add a guest physical address type as well.

However, this only makes sense if there is interest in adopting this approach. Otherwise, we'll just end up with a [XKCD927](https://xkcd.com/927/) situation. So, what are your thoughts?

Freax13, 2024-09-22:
Can you elaborate on what specific types/traits/etc would go into the common crate, what types/implementations would remain in this crate (or the other arch-specific crates), and how this would enable the reuse of code?

jounathaen, 2024-09-23:
Sure,
so I don't have a protoype for this yet, but my approach would be to copy-paste https://github.com/rust-osdev/x86_64/blob/master/src/addr.rs and add [the functions from the x86 crate](https://docs.rs/x86/0.52.0/x86/bits64/paging/struct.PAddr.html) so that it'd be a drop-in replacement for both crates. x86_64 could then simply replace _addr.rs_ with `pub use addresscrate::{VirtAddr, PhysAddr}` 

To illustrate code reuse and interoperability:
In HermitOS, we use numerous functions from the x86_64 crate, but as we target other architectures as well, we have to add our own physical/virtual address types. So anytime we call a function in the x86_64 crate, we have to do something like [`VirtAddress::new(addr.as_u64())`](https://github.com/hermit-os/kernel/blob/e608471f42d7b79f871fbe6d60a8b30e50043aef/src/arch/x86_64/mm/virtualmem.rs#L74).
Also, we are writing our own hypervisor and have defined the hypercalls and data layouts in a separate crate to reduce code duplication. [Parameters in the hypercalls are physical addresses](https://github.com/hermit-os/uhyve/blob/e5a38b2b5b9a4d4d01a3f807e335f9ebba329d4e/uhyve-interface/src/parameters.rs#L119), and it would be nice to use a type here that can seamlessly integrate with the x86_64 functions as well as the respective functionality for other architectures.

Freax13, 2024-09-23:
I'm not yet entirely clear how moving the address type alone helps reusability across architectures:
In this crate, the `VirtAddr` isn't dependent on the build-platform, we expose the same functions with the same x86_64-specific behavior even if the target platform isn't x86_64. To me, this implies that the x86_64-specific `VirtAddr` would have to remain a separate type and other types would be needed for other architectures. If there are different types for different platforms, I'm not sure how they could be reusable. What's your take on this?

jounathaen, 2024-09-23:
Ok, so what you want to say is that you also want the x86_64 crate to export the x86_64 address type (`u64`), even if it is compiled on some other architecture (like for example when programming an emulator), right?

That's a good point. I was thinking of something like this:

```rust
pub mod arch {
  pub mod x86_64 {
    #[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
    #[repr(transparent)]
    pub struct VirtAddr(u64);
    // ...
  }
  
  pub mod x86 {
    // Has the same functions (unless architecture specific) as arch::x86_64::VirtAddr.
    #[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
    #[repr(transparent)]
    pub struct VirtAddr(u32);
    // ...
  }
}

#[cfg(target_arch=x86_64)]
pub use crate::arch::x86_64::VirtAddr;

#[cfg(target_arch=x86)]
pub use crate::arch::x86::VirtAddr;

// Fallback type. Should probably not be used but prevents CI build/clippy errors when building for arch=none
#[cfg(not(any(target_arch = "x86_64", target_arch = "x86")))]
#[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
#[repr(transparent)]
pub struct VirtAddr(usize);
```

and the _x86_64_ crate would then `pub use addcrate::aarch::x86_64::VirtAddr`.

Freax13, 2024-09-23:
> Ok, so what you want to say is that you also want the x86_64 crate to export the x86_64 address type (`u64`), even if it is compiled on some other architecture (like for example when programming an emulator), right?

That's what we do right now, and we even have CI set up to test that.

> That's a good point. I was thinking of something like this:
> 
> ```rust
> pub mod arch {
>   pub mod x86_64 {
>     #[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
>     #[repr(transparent)]
>     pub struct VirtAddr(u64);
>     // ...
>   }
>   
>   pub mod x86 {
>     // Has the same functions (unless architecture specific) as arch::x86_64::VirtAddr.
>     #[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
>     #[repr(transparent)]
>     pub struct VirtAddr(u32);
>     // ...
>   }
> }
> 
> #[cfg(target_arch=x86_64)]
> pub use crate::arch::x86_64::VirtAddr;
> 
> #[cfg(target_arch=x86)]
> pub use crate::arch::x86::VirtAddr;
> 
> // Fallback type. Should probably not be used but prevents CI build/clippy errors when building for arch=none
> #[cfg(not(any(target_arch = "x86_64", target_arch = "x86")))]
> #[derive(Clone, Copy, PartialEq, Eq, PartialOrd, Ord, Hash)]
> #[repr(transparent)]
> pub struct VirtAddr(usize);
> ```
> 
> and the _x86_64_ crate would then `pub use addcrate::arch::x86_64::VirtAddr`.

That code structure wouldn't necessarily require the address type to be moved into the common crate though, would it? Couldn't you just re-export the crate from this type in the common crate?

jounathaen, 2024-09-23:
First, I think this contradicts the idea of a unification, as it makes it way harder to keep different architecture variants on-par.
Second, if the code is not part of that _common_ crate, it is not possible to add functions that are necessary to keep compatibility with other crates (take the [`split`](https://docs.rs/x86/0.52.0/x86/bits64/paging/struct.PAddr.html#method.split) function from x86 as a random example)

mkroening, 2024-09-23:
My approach to this (which I have not had the time to iron out yet) is similar to what @Freax13 has been describing:

The purpose of the “common” crate is to have types to differentiate between virtual and physical addresses in cross-architecture operating systems, while being able to easily interface with crates such as `x86_64`. This makes the types dependent on the compile target. The x86-64 crate explicitly uses `u64` everywhere to make its architecture-specific definitions usable from all compilation targets.

What I am thinking of would look something like this:

```rust
VirtAddr(usize);

impl From<crate::VirtAddr> for usize;
// More arch-independent helper functionality

#[cfg(all(feature = "x86_64", target_arch = "x86_64"))]
{
    impl From<crate::VirtAddr> for x86_64::VirtAddr;
    impl From<x86_64::VirtAddr> for crate::VirtAddr;
    // Checks for validity according to `x86_64::VirtAddr`
    impl TryFrom<usize> for crate::VirtAddr;
}
```

This keeps the invariants of the `x86_64` types intact while allowing easy cross-platform functionalities and interfacing with arch-independent abstractions that care about the kind of address as well as with arch-dependent abstractions from crates such as `x86_64`.

So I think such a type would be a bad fit for `x86_64` by design (wanting target-arch-dependence and thus wanting `usize` instead of `u64`, etc.), but a common type outside `x86_64` for arch-dependence may be useful while still allowing interoperating with arch-independent code such as `x86_64`.

Freax13, 2024-09-23:
> First, I think this contradicts the idea of a unification, as it makes it way harder to keep different architecture variants on-par. 

On the one hand, I get that keeping different crates in sync is difficult if not impossible, but on the other hand, if someone using the x86_64 crate wants to change the interface of an address type (e.g. add a method), I wouldn't want them to be forced to support all other architectures as well.

> Second, if the code is not part of that _common_ crate, it is not possible to add functions that are necessary to keep compatibility with other crates (take the [`split`](https://docs.rs/x86/0.52.0/x86/bits64/paging/struct.PAddr.html#method.split) function from x86 as a random example)

I'm not sure that this is a good argument though: I get why this would be very desirable for people targeting multiple platforms, but adding methods that don't necessarily make sense for one target just to be compatible with the method from another target doesn't feel right to me if you only care about one platform.

Here's another question: Do you care about methods working with const? If not, instead of moving all types into the *common* crate, can we have only traits + implementations +  a re-export of the target's address types live in the *common* crate? That way the common crate could define whatever methods desired including those that only exist to be compatible with other architecture, but none of the architecture-specific crates are forced to implement a specific interface.

mkroening, 2024-09-24:
> can we have only traits + implementations + a re-export of the target's address types live in the common crate

I don't think we can do without at least a fallback type for architectures without specific support-crate. At that point, I would personally lean towards a common type with easy conversion, such as the one I drafted above. I am not sure, though. To me, it feels like deciding between a few `.into()` calls and additional trait imports while also thinking about implementation complexity. Using `.into()` might be easier to extend for new architectures than having to implement all the methods via the trait again.

jounathaen, 2024-10-07:
Sorry for not responding earlier. I think that @mkroening s approach is a valid one without much disruption, but it also weakens the potential advantages. Anyway, I think I'll proceed locally in our project by creating something similar but without a global approach. We can come back to this later, if it'll provide value for others or there are other opinions on a "universal address type".

Feel free to close this issue.

closed 2024-10-12 by Freax13