rust-osdev/x86_64 #499: CI failure because of stabilized `const_mut_refs` feature

phil-opp, 2024-10-07:
The `const_mut_refs` feature was stabilized on nightly, which results in the following warning:

```
warning: the feature `const_mut_refs` has been stable since 1.83.0-nightly and no longer requires an attribute to enable
 --> src/lib.rs:5:43
  |
5 | #![cfg_attr(feature = "const_fn", feature(const_mut_refs))] // GDT::append()
```

On the CI we treat warnings as errors, which leads to build errors: https://github.com/rust-osdev/x86_64/actions/runs/11208838715/job/31153121364

There are multiple ways to handle this:

- Remove the `#![cfg_attr(feature = "const_fn", feature(const_mut_refs))]` feature gate. This will break people on older nightlies.
- Try to use `rustversion` if possible. I tried it, but I get an "inner macro attributes are unstable" error.
- Set the `stable_features` lint to `warn` on CI. This way, the warning is not turned into an error.

I'm leaning towards the third option, but I wanted to hear your opinions @josephlr and @Freax13.  

Once `const_mut_refs` is stablilized on stable, we can use `rustversion` to make the `GDT::append` function const on stable builds too. Then we can maybe deprecate the `const_fn` feature.

Freax13, 2024-10-07:
> The `const_mut_refs` feature was stabilized on nightly, which results in the following warning:
>
> ```
> warning: the feature `const_mut_refs` has been stable since 1.83.0-nightly and no longer requires [...]

I'm not sure that we can prevent breaking older nightlies in the long run. The problem is that it's fairly common for nightly features to be renamed or split into multiple smaller features. 

>   * Try to use `rustversion` if possible. I tried it, but I get an "inner macro attributes are unstable" error.

This might become feasible once [`cfg(version)`](https://doc.rust-lang.org/unstable-book/language-features/cfg-version.html) is stabilized.

>   * Set the `stable_features` lint to `warn` on CI. This way, the warning is not turned into an error.

That's fine with me.

> I'm leaning towards the third option, but I wanted to hear your opinions [@josephlr](https://github.com/josephlr) and [@Freax13](https://github.com/Freax13).
>
> Once `const_mut_refs` is stablilized on [...]

👍

phil-opp, 2024-10-07:
> I'm not sure that we can prevent breaking older nightlies in the long run. The problem is that it's fairly common for nightly features to be renamed or split into multiple smaller features.

I fully agree! We definitely shouldn't put any extra work into supporting older nightlies. People that pin specific nightly versions can also pin specific `x86_64` versions.

I just thought that setting `stable_features = warn` is not any extra work and it will probably spare us some work in the future. However, it will result in warnings, which are probably visible in downstream crates too. I'm not sure if this is preferable.

phil-opp, 2024-10-07:
I tried `-Dwarnings -Wstable-features` in https://github.com/rust-osdev/x86_64/pull/500, but it does not seem to work like I assumed. I'll open a PR to just remove the `const_mut_refs` feature instead.

closed 2024-10-08 by Freax13