rust-osdev/x86_64 #523: volatile in PageTable

mrjbom, 2024-12-07:
I see that the PageTable entry does not use volatile, is this a problem? I am afraid that the compiler will remove the values assigned to the PageTable.

Freax13, 2024-12-08:
The compiler can't always remove non-volatile accesses, it can only do so in specific situations. I don't think we've ever seen any problems related to the compiler removing page table accesses. This will also depend on how the references to the page tables are created in the first place. `OffsetPageTable` and `RecursivePageTable` both do some pointer math internally to create the references to the page tables and that should also make it more difficult for the compiler to optimize out accesses. 

Another related issue is that the CPU will atomically set dirty and accessed bits, so we technically shouldn't use mutable references to access the entries. I thought there was already an issue for this, but now I can't find it.

mrjbom, 2024-12-08:
Hopefully there won't be any problems related to this in the future.

closed 2024-12-08 by mrjbom