rust-osdev/x86_64 #534: `update_flags` does not add flags to parent page tables

ChocolateLoverRaj, 2025-02-05:
Steps to reproduce:
- Create a mapping without `USER_ACCESSIBLE`
- Update the flags to include `USER_ACCESSIBLE`

Expected:
The page can be accessed by user mode.

Actual:
The page can't be accessed by user mode because even though the L1 entry has `USER_ACCESSIBLE`, the L2, L3, L4 entries may not be user accessible.

Freax13, 2025-02-11:
Note that this isn't unique to `USER_ACCESSIBLE`, the same should be true for other bits e.g. `WRITE` and `NO_EXECUTE`.

I think we should be careful changing about `update_flags`'s behaviors around this, it's possible that some users depend on the current behavior, so I'd rather not change it. That said I don't think we currently offer a good solution to this, so I'm open to suggestions :)

ChocolateLoverRaj, 2025-02-11:
One idea is to set all the flags for all L4-L2 entries and only use L1 flags to control permissions.

Freax13, 2025-02-12:
I don't think we should change the behavior of existing methods in such a drastic way. There might be people that depend on the current behavior.

ChocolateLoverRaj, 2025-02-12:
What about if `update_flags` was left as is and there was a `update_flags_2` function with the new behavior?

Freax13, 2025-02-12:
I'm open to that idea (although I don't like that method name).

Another idea might be to improve `TranslateResult` to also contain the page table flags for the upper levels. With this information, a user can call `set_flags_p{2,3,4}_entry` as required. Changing `TranslateResult` would be a breaking change though...

phil-opp, 2025-02-13:
In the `Mapper::map_to` function, we automatically set the `PRESENT`, `WRITABLE` and `USER_ACCESSIBLE` flags for the parent table if required. We also have `Mapper::map_to_with_table_flags` with an explicit `parent_table_flags` argument. It probably makes sense to provide this functionality also for flag updates.

I'm also against changing the behavior of `update_flags` directly since this would be a very subtle breaking change, but a new method with a different name sounds good to me. Maybe something like `update_flags_recursive` or `update_flags_with_parents`?

For completeness, we could also add a method like `update_table_flags(page, flags, parent_table_flags)` that allows you to specify the flags for parent table entries manually.

---

Including the parent table flags in `TranslateResult` sounds like a good idea too! So maybe we just do both?

ChocolateLoverRaj, 2025-02-13:
I like the idea of a new function called `update_flags_with_parents`. I don't think I will be using the parent table flags so that change doesn't make a difference for my use.