rust-osdev/x86_64 #537: Adding and subtracting `VirtAddr` does not canonicalize when bit 47 changes

ChocolateLoverRaj, 2025-03-15:
There are two really simple way to reproduce this bug:
```rs
VirtAddr::new_truncate(0xFFFF_7FFF_FFFF_FFFF) + 1;
```
```rs
VirtAddr::new_truncate(0xFFFF_8000_0000_0000) - 1;
```
And this panics:
```
panicked at /home/rajas/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/x86_64-0.15.2/src/addr.rs:75:23:
virtual address must be sign extended in bits 48 to 64
```
I think this would be really easy to fix, just change `VirtAddr::new` to `VirtAddr::new_truncate` in the implementations of `Add` and `Sub`.

I ran into this error cuz my OS has kernel on higher half of usable address space and user space on lower half which is why my code involves `0xFFFF_8000_0000_0000`.

Freax13, 2025-03-15:
Both of those cases are intended behavior. We used to have this behavior im 0.14.x, but changed it with 0.15.x. If you want an add operation that skips the address gap, consider using the Step trait implementation (nightly only).

ChocolateLoverRaj, 2025-03-15:
Why is this the intended behavior?

Freax13, 2025-03-15:
IIRC, the main reasons are:
1. This behavior is consistent with normal integer addition: 0x7fff_ffff_ffff + 1 = 0x8000_0000_0000, not 0xffff_8000_0000_0000. 0x8000_0000_0000 is not a valid address, hence the panic. If you want to ask the question, "What's the next valid address?", that's literally what the Step trait is for.
2. A lot of kernels use the upper half for themselves and place userspace in the lower half. With that in mind, jumping the gap silently seems dangerous (although kernel should still check the result after addition).
3. Once we add support for 5-level paging (I'm planning to do so), the result of 0x7fff_ffff_ffff + 1 would change if addition silently did the sign extension: On 4 level paging this would result in 0xffff_8000_0000_0000, but on 5-levels this would result in 0xff00_0000_0000_0000. This seems rather unexpected.

ChocolateLoverRaj, 2025-03-15:
I think of a `VirtAddr` as just a 48 bit integer with the 12 other bits just needing to be a certain value. So it seems very strange that adding / subtracting within the 48 bits causes panics. Maybe this is the wrong way to think about it but idk why not.

For 2, this is what I did which caused the panic:
```rs
let temp_page =  Page::from_start_address(VirtAddr::new_truncate(KERNEL_VIRT_MEM_START)).unwrap()  - 1;
```
which finds the last page in user space to use as a temporary page for something.

So if this is the intended behavior then I will just use the `Step` trait. I hope your 5-level paging implementation also clarifies the behavior of adding / subtracting.

Freax13, 2025-03-15:
That's the intended behavior, and you'll want to use the Step trait. The addition of 5-level paging support will not change any of this.

While the addresses could be thought of as 48-bit values, we don't do this. We think of them as 64-bit values. `0xffff_8000_0000_0000 - 0x7fff_ffff_ffff` is well defined and will result in the value that would also be given by normal integer subtraction: 0xffff_0000_0000_0001, not 1, so that's also what you'd need to add when you wanted to jump the address gap.

There's nothing _wrong_ with thinking about addresses as 48-bit values, but we don't do this, and this is consistent across all VirtAddr methods. None of them will silently truncate/extend unless explicitly stated.

closed 2025-03-15 by Freax13