rust-osdev/x86_64 #541: VirtAddr subtraction panics on sign-extend boundary

tsatke, 2025-04-14:
```rust
    info!("before");
    let x = VirtAddr::new(0xFFFF_8000_0000_0000);
    info!("between");
    let y = x - 1;
    info!("after");
```

This panics in `panicked at /Users/tsatke/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/x86_64-0.15.2/src/addr.rs:75:23`, because `sub` just performs a `checked_sub` on `self.0`, without making sure that the resulting address is properly sign-extended.

Freax13, 2025-04-21:
That's the intended behavior. `Add`/`Sub` don't implicitly jump the gap in the address space, but `Step` does. `Step` and `Add`/`Sub` are different; the steps between two elements are not always equal to the numerical difference between them, so `step_forward`/`step_backward` are not equivalent to addition/subtraction. [`char`](https://doc.rust-lang.org/src/core/iter/range.rs.html#444) is another type where this is the case.

tsatke, 2025-04-22:
Got it. Maybe it makes sense to document that somewhere? Or even panic with a message mentioning that this is intended behavior for debug builds?

closed 2025-04-26 by Freax13

tsatke, 2025-05-01:
Thanks @Freax13 !

PR #543 (2025-04-26): improve `VirtAddr` `Add` & `Sub` impls
This PR improves the panic messages and documentation for `VirtAddr`'s `Add`/`Sub` impls.

Cc @tsatke
Closes #541