rust-osdev/x86_64 #553: Type mismatch compilation error on rustc 1.90.0-nightly

wenxuanjun, 2025-06-26:
In this rustc version (0fa4ec6cd 2025-06-25), the compiler assumes that a function with signature `extern ‘x86-interrupt’` should not never return (`!`):

```rust
error: invalid signature for `extern "x86-interrupt"` function
   --> interrupts.rs:107:82
    |
107 | ...terruptStackFrame, error_code: u64) -> ! {
    |                                     ^
    |
    = note: functions with the "custom" ABI cannot have a return type
help: remove the return type
   --> interrupts.rs:107:82
    |
107 | ...terruptStackFrame, error_code: u64) -> ! {
    |
```

However, `idt.double_fault` requires a `DivergingHandlerFuncWithErrCode` to be passed in:

```rust
error[E0308]: mismatched types
   --> interrupts.rs:46:29
    |
46  |             .set_handler_fn(double_fault)
    |              -------------- ^^^^^^^^^^^^ expected fn pointer, found fn item
    |              |
    |              arguments to this method are incorrect
    |
    = note: expected fn pointer `extern "x86-interrupt" fn(x86_64::structures::idt::InterruptStackFrame, _) -> !`
                  found fn item `extern "x86-interrupt" fn(x86_64::structures::idt::InterruptStackFrame, _) -> () {arch::interrupts::double_fault}`
```

workingjubilee, 2025-06-27:
working on a fix.

fangpin, 2025-07-29:
fixed at https://github.com/rust-osdev/x86_64/pull/560

asahi-myzk, 2025-08-16:
This is finally fixed in the latest nightly version (cd7cbe818 2025-08-15).

closed 2025-08-16 by Freax13