rust-osdev/x86_64 #557: An error about the `set_general_handler` macro

zhangxuan2011, 2025-07-12:
I was making a kernel recently. But my VScode give me this error suddenly:

```error
error: invalid signature for `extern "x86-interrupt"` function
  --> kernel/src/interrupts/idt.rs:11:9
   |
11 |         set_general_handler!(&mut idt, general_handler);
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |
   = note: functions with the "custom" ABI cannot have a return type
help: remove the return type
  --> kernel/src/interrupts/idt.rs:11:9
   |
11 |         set_general_handler!(&mut idt, general_handler);
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: this error originates in the macro `$crate::set_general_handler_entry` which comes from the expansion of the macro `set_general_handler` (in Nightly builds, run with -Z macro-backtrace for more info)

```

This is the code:

`src/interrupts/idt.rs`:
```rust
use crate::interrupts::error_handler::general_handler;
use lazy_static::lazy_static;
use x86_64::set_general_handler;
use x86_64::structures::idt::InterruptDescriptorTable;

// Use `lazy_static` to delay initialization
lazy_static! {
    pub static ref IDT: InterruptDescriptorTable = {
        let mut idt = InterruptDescriptorTable::new();
        // The general handler
        set_general_handler!(&mut idt, general_handler);
        idt
    };
}

// Initialize and load IDT
pub fn init_idt() {
    IDT.load();
}
```

`src/interrupts/error_handler.rs`:
```rust
use crate::serial_println;
use x86_64::structures::idt::InterruptStackFrame;

pub fn general_handler(
   stack_frame: InterruptStackFrame,
   index: u8,
   error_code: Option<u64>,
) {
    serial_println!("EXCEPTION: INDEX {}, \n{:#?};\nerror code: {:?}", index, stack_frame, error_code);
}

```

I don't know what happened exactly, and im only 14 so im not so clever :( . Please, this is important!

Freax13, 2025-07-12:
This error appears starting with Rust nightly 2025-06-25. There are plans to fix this eventually, but that work hasn't been completed yet. Until then, please downgrade to Rust nightly 2025-06-24 (or earlier).

zhangxuan2011, 2025-07-13:
> This error appears starting with Rust nightly 2025-06-25. There are plans to fix this eventually, but that work hasn't been completed yet. Until then, please downgrade to Rust nightly 2025-06-24 (or earlier).

So this is the problem of Rust compiler? After fixed it, please publish it as soon as possible, thanks!